sparkfun/SparkFun_Simultaneous_RFID_Tag_Reader_Library #2: Module stops responding

TechieRuss, 2017-03-08:
I am by no means an expert, but after fussing with this thing for a few days I think I nailed down the problem and a possible solution. I will try to explain. (again. I am no expert and what I write here will reflect this.)

If anyone who is an expert can confirm this, suggest a better solution, or/and re-write this in a better/clearer way.... PLEASE!

When you reset the arduino without disconnecting power from the entire assembly (like when you update firmware) or if it "browns out" during a read, the module locks up and does not respond to commands. 
From what I found in the datasheets and from by playing with their autonomous config tool this is a similar issue to when the module is in autonomous mode and needs to be reset. On thingmagic's other modules there is a reset pin to help with this, the nano does not. 
At first I was stumped as to wtf could help with this. And almost began to pullout my hair. Then I remembered something in the datasheets about a bootloader.  When the module is powered on the bootloader needs a little time(100-200ms) to do it's thing. After that the "lockedup" state is once again a pain in the fleshy parts used for sitting by humans. If you manage to run the setupNano() within the first ~200ms the module can be reset. I, however, had a long delay() to show a "loading.." message on an LCD before running it.(irony?)
Removing the delay let me reset the module after it became unresponsive by removing power from the entire contraption. This however did not help when I had my code try to reconfigure the module as part of the proccess. 

My solution was to wire up a digital pin to the EN pin on the breakout writing a low for 200ms then a high to start it up again and calling the setup routine. this seems to have worked.

nseidle, 2018-03-20:
Hi TechieRuss, I can't replicate your setup but it sounds like a great solution to your problem. I'll leave open.

Tweeks-va, 2020-05-10:
How can we (through code on the ARduino) simply change the duty cycle? That's the recommended way to prevent this issue?