sparkfun/SparkFun_Simultaneous_RFID_Tag_Reader_Library #4: Set Power Mode?

JohnRusk, 2017-07-12:
The documentation for the ThingMagic module describes various Power Modes.  They look like an ideal solution for managing thermal issues, especially in cases like mine where i'll be needing to read only occasionally.

But I can't find any way to set the Power Mode via the Arduino API.  Can you suggest any solution/workaround that would allow me to set the Power Mode?

lewispg228, 2017-07-12:
In production we test these at 10 dBm, and this is the command we use from the test fixture:

`nano.setReadPower(1000); //10.00 dBm.`

Hope this helps!

JohnRusk, 2017-07-12:
That's different from what I'm asking about.  that's "power level". "power
mode" tells it to sleep,  ie shut down most of its circuits,  when bit
actively reading.

JohnRusk, 2017-07-12:
Just corrected a typo in last sentence of previous comment.

lewispg228, 2017-07-12:
@JohnRusk, sorry about that confusion. We don't adjust the power mode in production, and I don't believe the current library has that control included. You might want to try posting a comment on our product webpage. (  https://www.sparkfun.com/products/14066  ) There is a lot of action in the comments there and someone else may be able to help find a solution. Good luck!

TechieRuss, 2017-07-12:
I'm using the enable pin to do this. Connect the enable pin to any gpio pin on the arduino and write low/high to disable/enable.

JohnRusk, 2017-07-12:
Thanks @lewispg228 and @TechieRuss for your prompt replies, and also for the confirmation of how to use the enable pin instead. I suspect that's what I'll do.  Just requires re-initialization after enablement, because the enable is basically a reboot of the reader but, in the absence of API support for Power Mode, I can live with the reboots.

closed 2017-07-12 by JohnRusk