sparkfun/SparkFun_Simultaneous_RFID_Tag_Reader_Library #6: Cant compile expamles

kartikye, 2017-10-27:
D:\Documents\Arduino\libraries\SparkFun_Simultaneous_RFID_Tag_Reader_Library\src\SparkFun_UHF_RFID_Reader.cpp: In member function 'uint8_t RFID::writeTagEPC(char*, uint8_t, uint16_t)':

D:\Documents\Arduino\libraries\SparkFun_Simultaneous_RFID_Tag_Reader_Library\src\SparkFun_UHF_RFID_Reader.cpp:277:63: error: invalid conversion from 'char*' to 'uint8_t* {aka unsigned char*}' [-fpermissive]

   return (writeData(bank, address, newID, newIDLength, timeOut));

                                                               ^

In file included from D:\Documents\Arduino\libraries\SparkFun_Simultaneous_RFID_Tag_Reader_Library\src\SparkFun_UHF_RFID_Reader.cpp:42:0:

D:\Documents\Arduino\libraries\SparkFun_Simultaneous_RFID_Tag_Reader_Library\src\SparkFun_UHF_RFID_Reader.h:119:10: note: initializing argument 3 of 'uint8_t RFID::writeData(uint8_t, uint32_t, uint8_t*, uint8_t, uint16_t)'

  uint8_t writeData(uint8_t bank, uint32_t address, uint8_t *dataToRecord, uint8_t dataLengthToRecord, uint16_t timeOut = COMMAND_TIME_OUT);

          ^

exit status 1
Error compiling for board Arduino/Genuino Uno.

quitmeyer, 2017-12-02:
I had the same error, i just forgot to install the Simultaneous RFID library on this computer's arduino

kdoyle35, 2018-03-15:
Arduino: 1.6.5 (Windows XP), Board: "Arduino/Genuino Uno"
Build options changed, rebuilding all

C:\Documents and Settings\kdoyle\My Documents\Arduino\libraries\SparkFun_Simultaneous_RFID_Tag_Reader_Library\src\SparkFun_UHF_RFID_Reader.cpp: In member function 'uint8_t RFID::writeTagEPC(char*, uint8_t, uint16_t)':
C:\Documents and Settings\kdoyle\My Documents\Arduino\libraries\SparkFun_Simultaneous_RFID_Tag_Reader_Library\src\SparkFun_UHF_RFID_Reader.cpp:277:63: error: invalid conversion from 'char*' to 'uint8_t* {aka unsigned char*}' [-fpermissive]
   return (writeData(bank, address, newID, newIDLength, timeOut));
                                                               ^
In file included from C:\Documents and Settings\kdoyle\My Documents\Arduino\libraries\SparkFun_Simultaneous_RFID_Tag_Reader_Library\src\SparkFun_UHF_RFID_Reader.cpp:42:0:
C:\Documents and Settings\kdoyle\My Documents\Arduino\libraries\SparkFun_Simultaneous_RFID_Tag_Reader_Library\src\SparkFun_UHF_RFID_Reader.h:119:10: error:   initializing argument 3 of 'uint8_t RFID::writeData(uint8_t, uint32_t, uint8_t*, uint8_t, uint16_t)' [-fpermissive]
  uint8_t writeData(uint8_t bank, uint32_t address, uint8_t *dataToRecord, uint8_t dataLengthToRecord, uint16_t timeOut = COMMAND_TIME_OUT);

        ^
Error compiling.

*****************************************************************************************************
I have installed the Simultaneous RFID library but I still get the error.....please help!

chelofagalde, 2018-03-18:
Hey guys, getting the same error here. Have you guys been able to solve it?

kdoyle35, 2018-03-18:
Hi. 
Actually, in the end I upgraded my Arduino IDE version from 1.6.5 to 1.8.5 and that solved it!!

I got the code (Example 1_Constant_Read) to compile and upload to my Arduino Uno, but now I'm getting a message telling me:
"Module failed to respond. Please check wiring."   

Looks like it can't configure the nano to run at 38400 baud??

kdoyle35, 2018-03-19:
I have a Sparkfun Simultaneous RFID Reader connected as a shield to an Arduino Uno. I'm running Arduino IDE version 1.8.5
I have just run Example 1 (Constant Read) and Example 2 (Read EPC) and both are failing to read any tags. I am using UHF RFID Tags (WRL-14147 recommended by Sparkfun).....

see attached output from Serial Monitor below:

[image]

Anyone have the same issue? Can anyone help me out here please? It would be much appreciated.

nseidle, 2018-03-20:
We're mixing issues. Please open a separate issue.

A couple background things going on: I'm able to compile the examples but it looks like the Arduino library manager is not picking up the latest v1.0.1 of the library. I will verify everything and then roll the lib ver forward again.

I'm closing this issue as it seems solved (lib doesn't work in v1.6 and or user didn't have lib installed).

closed 2018-03-20 by nseidle