sparkfun/SparkFun_Simultaneous_RFID_Tag_Reader_Library #7: Doesn't build: line 676

easterbilby, 2018-03-25:
Currently line 676 reads:

else if (this.msg[1] == 0x0a) //temperature

This is throwing an error. The fix seemed to be easy - I removed the "this", per the prior code:

else if (msg[1] == 0x0a) //temperature

nseidle, 2018-03-25:
That does look like a vestige. What platform are you compiling on and for?

easterbilby, 2018-03-28:
Windows 10, Arduino 1.8.5  for the Nano. There were a couple of warnings that turned up when compiling, but they didn't create any runtime issues, and the one error.

nseidle, 2018-03-31:
Looks like the bug snuck in [this PR](https://github.com/sparkfun/SparkFun_Simultaneous_RFID_Tag_Reader_Library/pull/3). Bug fixed. New ver released. Please pull down from arduino lib manager when possible.

*Thank you* for reporting this.

closed 2018-03-31 by nseidle