sparkfun/SparkFun_Simultaneous_RFID_Tag_Reader_Library #8: TID length inconsistency

billvoss, 2018-04-18:
Hi,

I'm having an issue where the 12th byte of the tag TID randomly changes on each read.

Perhaps related - I noticed that the code says that TIDs are 20 bytes (line 10, Example9_Read_TID.ino), but the product page (and ThingMagic) say that the TIDs are 64 bits (https://www.sparkfun.com/products/14151).

Do you know what's going on here?

Thanks

nseidle, 2018-04-18:
I've been digging into this. I think the short answer is that most tag TIDs are 64 bit or 8 byte. The long answer is that it can be variable.

Here's the Gen2 Spec: https://www.gs1.org/sites/default/files/docs/epc/GS1_EPC_TDS_i1_10.pdf

See section 16.2.2 XTID Serialization. There's room for a variable length TID.

I'm going to keep testing and looking for a tag that has more than 8 bytes. In the mean time, you may want to adjust your code so it only looks at the first 8 bytes returns from the TID function call.

nseidle, 2018-04-18:
To clarify: The minimum TID is 24 bit and contains things like the tag MFG and Model:

[image]

What this issue is discussing is the Unique ID or UID portion within the extended TID section:

[image]

As seen above they are rare but 20byte UIDs do exist.

[image]

I have found many more 8 byte UID.

nseidle, 2018-04-18:
.readTID() is a bad name for a function because it's really reading the UID. So I added a .readUID() function in the latest commit. I left .readTID() in place in case users have code that use it.

I was getting time outs when the sketch was looking for a 20 byte UID but a tag only had a 8 byte UID. So I have re-written the example, and cleaned up a bunch of the confusing language. It's now called "Read_UID" and it's in the advanced examples folder.

@billvoss - can you download the latest commit and give the UID example a try? If it works well for you then I'll do a release and the Arduino library stuff to get the new version to show up in the lib manager.

nseidle, 2018-04-19:
No problem at all! This is the software repo. The other is the hardware. I love closing issues so duplicates are fine with me ;)

billvoss, 2018-04-20:
Glad I could help :P 

I've tried the latest commit and the UID example works perfectly with the 30 tags I have. 

As an aside, I noticed that the first byte was always 01 and the forth is 00 - perhaps related to a particular batch of tags, or is further model specific information.

Thanks for looking into this so quickly Nate!

Bill

nseidle, 2018-04-20:
Excellent! Thanks for kicking the tires.

Are the tags all the same make and model? They may be serialized (at least on the mfg line) which would explain the 01 up front. Not sure about the 0x00 at the end. I'm able to read different UIDs from a grab bag of different tags. I'll close this issue out and push a release. Please bring up any other issues you may have.

closed 2018-04-20 by nseidle

billvoss, 2018-04-21:
Cheers! They are the all the same make and model, which would make sense.