sparkfun/SparkFun_Simultaneous_RFID_Tag_Reader_Library #10: Using other options for startReading()

Kyaels, 2018-06-08:
I'm currently working with the SRTR and I want to be able to constant read the TID. I dug a little into the library, and noticed that the startReading() function has a blob that is sent to the module. I don't really know how the blob works. I've been looking through URA and running transport logs, but it doesn't make much sense to me. Can you explain the meaning behind the blob and how I can modify it to read TID? Thanks!

nseidle, 2018-06-08:
The blob was found by inspecting the serial stream between a SRTR and the [Universal Reader Assistant software](https://learn.sparkfun.com/tutorials/simultaneous-rfid-tag-reader-hookup-guide/using-the-universal-reader-assistant). Armed with that info, and digging to the API in C from ThingMagic, I found [the lines](https://github.com/sparkfun/SparkFun_Simultaneous_RFID_Tag_Reader_Library/blob/master/src/SparkFun_UHF_RFID_Reader.cpp#L98) that you see commented in the lib.

So you can kind of see what each byte does. For example, byte 6 is the protocol ID. We could spend a fair amount of time trying to back out what each byte what responsible for but it was more pertinent to get the overall lib working.

'TID' is often mistakenly used. Do you mean UID? Have you seen the advanced example [Read_UID](https://github.com/sparkfun/SparkFun_Simultaneous_RFID_Tag_Reader_Library/blob/master/examples/Advanced/Read_UID/Read_UID.ino)?

Kyaels, 2018-06-10:
Specifically, I'm looking for a way to read the UID* in the same manner that the EPC is read in the constant_read example. I was looking to do this by making modifications to the blob/startReading() function. That's why I wanted a bit more info on how the blob was deciphered.

nseidle, 2018-06-12:
It looks possible from the URA and some of the ThingMagic docs. If you can get the behavior you want via the URA then I can bake those serial commands into the library.

closed 2018-08-24 by Kyaels