sparkfun/SparkFun_Simultaneous_RFID_Tag_Reader_Library #12: Send a standard Class−1 Generation−2 UHF Select command

Holy1990, 2018-08-01:
I need to read the temperature from my RFID temperature sensor. For reading the temperature, 2 step process requiring standard UHF Select and Read commands.

Send a standard Class−1 Generation−2 UHF Select command with the parameters described in Table 6 to initialize the temperature sensor and calculate a Temperature Code. The Select command to be followed by 2 ms of continuous wave before the reader issues any further commands. This pause
gives the temperature sensor circuit time to run. The reader must not power down at any time between the Select and Read commands.

Send a standard Class−1 Generation−2 UHF Read command to retrieve the Temperature Code from the sensor tag memory at the location given in Table 7.

Is this SparkFun Simultaneous RFID Tag Reader Library for Arduino support to send the Select command? i only able to send the read command follow the example provided by sparkfun on Arduino.

nseidle, 2018-08-01:
We've seen a few RFID sensors, like humidity and temperature, but we haven't been successful in reading them.

Your issue talks about Tables XX. Can you link to them?

I don't think it's possible but I'll put out the question to ThingMagic.

nseidle, 2018-08-02:
Jadak's response is below. It looks like it may be possible. I do not plan to add this feature to the library at this time but if you or anyone else wants to propose a PR, please do so.

...

Please find the attached latest Nano firmware and Mercury API at https://www.jadaktech.com/wp-content/uploads/2018/06/mercuryapi-1.31.0.33.zip 

You can configure the T4 parameter and read the desired sensor value from the sensor tags. Please use the below code snippet to Paramget and Paramset the T4 parameter and let us know if you observe any issues. For more information on T4 parameter, please refer “Configurable T4 for Gen2 protocol” section(page 23) of the Nano Release Notes shared earlier(by Harinath).

UInt32 T4Value = 3000;
r.ParamSet("/reader/gen2/t4", T4Value);
T4Value = (UInt32)r.ParamGet("/reader/gen2/t4");
Console.WriteLine("Get new T4value from module : {0}", T4Value);

Below is the code snippet for fetching temperature after setting T4 parameter

filter = new Gen2.Select(false, Gen2.Bank.USER, 0xE0, 0, new byte[]{});
Gen2.ReadData rData = new Gen2.ReadData(Gen2.Bank.RESERVED, 0x0E, wordCount);

You can modify the code to fetch the On chip RSSI code and Sensor code by providing the appropriate data. 

We have also implemented Read after Write feature, for more information on this feature, please refer "Ability to “Read Data” immediately after sending a “Write EPC” or “Write Data” command" section(page 24) of Nano Release Notes. Write.cs sample codelet demonstrates this feature.

Holy1990, 2018-08-03:
Hi, Thank for your information... But i have try run the code set the T4Value to 3000, but it will return error (Specified cast is not valid) when "r.ParamSet("/reader/gen2/t4", T4Value);"

Steven-SS, 2019-05-07:
In [SparkFun_Simultaneous_RFID_Tag_Reader_Library/src/SparkFun_UHF_RFID_Reader.cpp](url) where it shows all the raw code for many of the functions. Is there any way to get the raw code for the Gen2.Select() function?? In order to retrieve the "C" value from the reserved bank. 
Not the "Gen2.Select(boolean invert, int bank, int bitpointer, int bitlength, byte mask)" but the actual raw code that is within that Select() function.
Thanks.

simplearetenir, 2019-05-24:
Hi,
Thank you for your library, the possibility of obtaining the temperature of the module would be really great. Can anyone give me a way to do that?
Regards

AC11115, 2021-01-20:
Hi All,

Did anyone have any luck with this? The temp sensing tag is a really nice little device.

ON semisconductor also have a moisture sensing tag which I've been able to get working with the M6e nano reader as it does not require a SELECT command. You send a READ command to a particular location in the reserved memory bank and you get the moisture reading.

Would be great to get the temp sensing patch working with the M6e nano too but I can't get the SELECT command working.

Thanks,

AC11115, 2021-01-22:
Hi All,

I'd like to add that if it's a temperature sensing tag you're after I've gotten the EM4325 transponder from EM microelectronic working with my sparkfun Simultaneous RFID reader.

To request a temperature reading you write to a certain memory location in the USER membank. No select command required. 

Not sure if anyone is selling these in tag form though.

https://www.emmicroelectronic.com/product/epc-and-uhf-ics/em4325

dyones7, 2023-11-14:
Hello,

This thread is a few years old, but I will still ask.

I wanted to check if anyone has had success with getting the recommended Jadak patches running or if anyone know if someone has implemented the SELECT command so that the Magnus S3 rfid based sensor tags temperature can be read.

Many Thanks