sparkfun/SparkFun_Simultaneous_RFID_Tag_Reader_Library #13: compile with Arduino MKR1000

trineon89, 2018-08-29:
Trying to get this library running with an [Arduino MKR1000](https://store.arduino.cc/arduino-mkr1000) but it fails with the following errors:
```C:\Users\X\Documents\Arduino\libraries\SparkFun_Simultaneous_RFID_Tag_Reader_Library\src\SparkFun_UHF_RFID_Reader.cpp: In member function 'uint8_t RFID::writeTagEPC(char*, uint8_t, uint16_t)':

C:\Users\X\Documents\Arduino\libraries\SparkFun_Simultaneous_RFID_Tag_Reader_Library\src\SparkFun_UHF_RFID_Reader.cpp:277:63: error: invalid conversion from 'char*' to 'uint8_t* {aka unsigned char*}' [-fpermissive]

   return (writeData(bank, address, newID, newIDLength, timeOut));

                                                               ^

In file included from C:\Users\X\Documents\Arduino\libraries\SparkFun_Simultaneous_RFID_Tag_Reader_Library\src\SparkFun_UHF_RFID_Reader.cpp:42:0:

C:\Users\X\Documents\Arduino\libraries\SparkFun_Simultaneous_RFID_Tag_Reader_Library\src\SparkFun_UHF_RFID_Reader.h:119:10: error:   initializing argument 3 of 'uint8_t RFID::writeData(uint8_t, uint32_t, uint8_t*, uint8_t, uint16_t)' [-fpermissive]

  uint8_t writeData(uint8_t bank, uint32_t address, uint8_t *dataToRecord, uint8_t dataLengthToRecord, uint16_t timeOut = COMMAND_TIME_OUT);

          ^

exit status 1
Error compiling for board Arduino/Genuino MKR1000.
```
Even if the Sketch is complete empty, the line `#include "SparkFun_UHF_RFID_Reader.h"` steps into this failure.

Some sort of workaround or even better a fix?
Thanks in advance

trineon89, 2018-10-04:
*push*

D3w3y, 2018-10-24:
I have a very dirty workaround for you. For me the compilation works with that, but I don't need this method so I don't care if it still works as expected or not.

In the `SparkFun_UHF_RFID_Reader.cpp` file add a line to the `writeTagEPC` method, to convert the `newID` from `char` to `uint8` and then hand over the new variable to the `writeData` method.

`writeTagEPC` looks like this for me now:
```
uint8_t RFID::writeTagEPC(char *newID, uint8_t newIDLength, uint16_t timeOut)
{
  uint8_t bank = 0x01; //EPC memory
  uint8_t address = 0x02; //EPC starts at spot 4
  uint8_t * nnewID = (uint8_t*)atoi(newID);   // New Line doing the conversion

  return (writeData(bank, address, nnewID, newIDLength, timeOut));   // Changed newID to the above created variable nnewID
}
```

For me this looks like a bug in the library, maybe @nseidle can have a look and fix it properly?

nseidle, 2018-10-24:
@D3w3y  - Thanks! Your code fixed the error. The Uno compiler settings are more forgiving than the MKR and Teensy settings.

Hi @trineon89 - I just committed D3w3y's fix, as well as two additional examples showing how to use hardware serial (instead of softwareSerial). I don't have a MKR1000 to test with but checkout the Examples->Advanced->Example3_HardwareSerial_Write_EPC. It now compiles and correctly writes an EPC to a tag using a Teensy.

I'll lib ver roll the library here shortly. Please test and let us know how it goes.

closed 2018-10-24 by nseidle

nseidle, 2019-04-09:
To document for myself: the atoi recommendation caused a bug in EPC write as reported by [ombi](https://github.com/sparkfun/SparkFun_Simultaneous_RFID_Tag_Reader_Library/issues/17) and fixed by [paulvha](https://forum.sparkfun.com/viewtopic.php?f=118&t=49574&p=203195#p203195). The latest v1.0.7 should fix both the bug and the MKR1000 compilation issue.