sparkfun/SparkFun_Simultaneous_RFID_Tag_Reader_Library #15: Reading tag EPC in ASCII code is possible?

sivabharaniii, 2018-12-18:
I was able to read EPC and user data in hexa,decimal,octa,binary.But i can't able to read it in ASCII,is it possible to read in ASCII.

nseidle, 2018-12-18:
Hi @sivebharaniii - Sounds like a code issue. If you can read in hex then you simple need to convert for your intended purposes (printing, comparing, or other can all be done in ASCII or HEX, it's just a matter of code). You are welcome to post here and I can try to trouble shoot but this is not an issue with the library so I'm going to close it out.

closed 2018-12-18 by nseidle

sivabharaniii, 2018-12-20:
No actually code is working fine while executing,only special ASCII characters are printing like ? and @.My issue is i cant able to print alphabet characters