sparkfun/SparkFun_Simultaneous_RFID_Tag_Reader_Library #17: Problem using EPCwrite

ombi, 2019-03-19:


ombi, 2019-03-19:
I fixed the problem that was generated by the last release of the library.  It introduces an incorrect atoi function at line 276.  The use of the atoi function on the twelve byte of EPC make no sense as it is a binary sequence.  As soon as the function is removed, it starts working properly again.

closed 2019-03-19 by ombi

nseidle, 2019-04-09:
Hi Ombi - Thank you for reporting the error and fix. I apologize, it went unnoticed until today because you closed the issue. I really appreciate your help tho. In the future, please keep issues open (until I can get them fixed :) ) or if you can create a PR then I'll be *really* ecstatic.

All the best,
-Nathan

nseidle, 2019-04-09:
Bug should be fixed in latest library release v1.0.7.