sparkfun/SparkFun_Simultaneous_RFID_Tag_Reader_Library #22: Module failed to respond. Please check wiring. ESP32 Serialsoftware

hicham-vv, 2021-04-01:
Hello, i am trying to communicate the module nano m6e with the esp32 via the software serial, i installed the ESPsoftwareSerial plerup/EspSoftwareSerial@^6.12.1.
i used the example1 "constant _read", but the module failed to respond, **Module failed to respond. Please check wiring.**
i checked all the wires but everything is allright, 

NB :_i used the Redboard Arduino, and i can read tags so everthing is good ,_ 

Any help, thanks

nseidle, 2021-04-01:
I avoid using software serial whenever possible. ESP32 has multiple hardware serial ports. Try using the hardware serial ports and let us know if you continue to run into issues.

closed 2021-04-01 by nseidle

hicham-vv, 2021-04-01:
All my hardware serial ports is occupied, any way thank you, i fixed the problem