sparkfun/SparkFun_Simultaneous_RFID_Tag_Reader_Library #25: Compile Errors -

JonGordon123, 2022-03-28:
Board: Sparkfun ESP32 Thing Plus
Arduino: 1.8.19
Library: 1.0.7
Example Code: Example1_HardwareSerial_Constant_Read.ino (Using Serial2 instead of Serial5)

> C:\Users\Jon\Documents\Arduino\libraries\SparkFun_Simultaneous_RFID_Tag_Reader_Library\src\SparkFun_UHF_RFID_Reader.cpp: In member function 'bool RFID::begin(Stream&)':
> C:\Users\Jon\Documents\Arduino\libraries\SparkFun_Simultaneous_RFID_Tag_Reader_Library\src\SparkFun_UHF_RFID_Reader.cpp:55:1: error: no return statement in function returning non-void [-Werror=return-type]
>  }
>  ^
> 
> C:\Users\Jon\Documents\Arduino\libraries\SparkFun_Simultaneous_RFID_Tag_Reader_Library\src\SparkFun_UHF_RFID_Reader.cpp: In member function 'void RFID::printMessageArray()':
> C:\Users\Jon\Documents\Arduino\libraries\SparkFun_Simultaneous_RFID_Tag_Reader_Library\src\SparkFun_UHF_RFID_Reader.cpp:847:20: warning: comparison is always false due to limited range of data type [-Wtype-limits]
>      if (amtToPrint > MAX_MSG_SIZE)
>                     ^
> 
> C:\Users\Jon\Documents\Arduino\libraries\SparkFun_Simultaneous_RFID_Tag_Reader_Library\src\SparkFun_UHF_RFID_Reader.cpp: In member function 'uint8_t RFID::parseResponse()':
> C:\Users\Jon\Documents\Arduino\libraries\SparkFun_Simultaneous_RFID_Tag_Reader_Library\src\SparkFun_UHF_RFID_Reader.cpp:710:1: error: control reaches end of non-void function [-Werror=return-type]
>  }
>  ^
> 
> cc1plus.exe: some warnings being treated as errors
> exit status 1
> Error compiling for board SparkFun ESP32 Thing Plus.
> 

**Made the following changes in SparkFun_UHF_RFID_Reader.cpp, and the compiler stopped complaining.**

Added "return true;" after line 52 (not sure if this is the best idea, but it works.... not sure what it is actually supposed to be returning)
```
//Initialize the Serial port
bool RFID::begin(Stream &serialPort)
{
  _nanoSerial = &serialPort; //Grab which port the user wants us to use
  return true;
  //_nanoSerial->begin(); //Stream has no .begin() so the user has to do a whateverSerial.begin(xxxx); from setup()
}
```

and, added "return (RESPONSE_IS_UNKNOWN);" after line 709

```
  //Check the CRC on this response
  uint16_t messageCRC = calculateCRC(&msg[1], msgLength - 3); //Ignore header (start spot 1), remove 3 bytes (header + 2 CRC)
  if ((msg[msgLength - 2] != (messageCRC >> 8)) || (msg[msgLength - 1] != (messageCRC & 0xFF)))
  {
    return (ERROR_CORRUPT_RESPONSE);
  }

  if (opCode == TMR_SR_OPCODE_READ_TAG_ID_MULTIPLE) //opCode = 0x22
  {
    //Based on the record length identify if this is a tag record, a temperature sensor record, or a keep-alive?
    if (msg[1] == 0x00) //Keep alive
    {
      //We have a Read cycle reset/keep-alive message
      //Sent once per second
      uint16_t statusMsg = 0;
      for (uint8_t x = 0; x < 2; x++)
        statusMsg |= (uint32_t)msg[3 + x] << (8 * (1 - x));

      if (statusMsg == 0x0400)
      {
        return (RESPONSE_IS_KEEPALIVE);
      }
      else if (statusMsg == 0x0504)
      {
        return (RESPONSE_IS_TEMPTHROTTLE);
      }
    }
    else if (msg[1] == 0x08) //Unknown
    {
      return (RESPONSE_IS_UNKNOWN);
    }
    else if (msg[1] == 0x0a) //temperature
    {
      return (RESPONSE_IS_TEMPERATURE);
    }
    else //Full tag record
    {
      //This is a full tag response
      //User can now pull out RSSI, frequency of tag, timestamp, EPC, Protocol control bits, EPC CRC, CRC
      return (RESPONSE_IS_TAGFOUND);
    }
  }
  else
  {
    if (_printDebug == true)
    {
      _debugSerial->print(F("Unknown opcode in response: 0x"));
      _debugSerial->println(opCode, HEX);
    }
    return (ERROR_UNKNOWN_OPCODE);
  }
  return (RESPONSE_IS_UNKNOWN);
}
```

asendotsinski, 2023-03-11:
I disagree.

This is obviously not an issue with the SoftwareSerial library but an issue of return types. Both of these functions are non-void, yet they have an execution path with *no return value*. This, as per the C++ standard, results in undefined behavior, which only incidentally does not crash most embedded boards.

It's possible Jon's interpretation of what should be returned is wrong - I'm not sure of the intent behind some of these return values either - but something needs to be returned. Otherwise, `RFID::begin()` might as well be `void`.

santaimpersonator, 2023-03-14:
@asendotsinski Nice catch, thanks for pointing that issue as well. The return mismatch is might be new, I could have sworn that the serial libraries used to return a `bool`.

---

For now, I would recommend users follow the example hardware setup in the hookup guide to avoid potential logic-level and power issues. The example still seems to compile with the Arduino AVR core *(i.e. for boards like the Arduino Uno)*. 

```
Compiling libraries...
Compiling library "SparkFun Simultaneous RFID Tag Reader Library"
"C:\\Users\\<username>\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Users\\<username>\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Users\\<username>\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\standard" "-Ic:\\Users\\<username>\\Documents\\Arduino\\libraries\\SparkFun_Simultaneous_RFID_Tag_Reader_Library\\src" "c:\\Users\\<username>\\Documents\\Arduino\\libraries\\SparkFun_Simultaneous_RFID_Tag_Reader_Library\\src\\SparkFun_UHF_RFID_Reader.cpp" -o "C:\\Users\\<username>\\AppData\\Local\\Temp\\arduino-sketch-0AC93B035618789504040DCC686C1B6A\\libraries\\SparkFun_Simultaneous_RFID_Tag_Reader_Library\\SparkFun_UHF_RFID_Reader.cpp.o"
c:\Users\<username>\Documents\Arduino\libraries\SparkFun_Simultaneous_RFID_Tag_Reader_Library\src\SparkFun_UHF_RFID_Reader.cpp: In member function 'bool RFID::begin(Stream&)':
c:\Users\<username>\Documents\Arduino\libraries\SparkFun_Simultaneous_RFID_Tag_Reader_Library\src\SparkFun_UHF_RFID_Reader.cpp:55:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
c:\Users\<username>\Documents\Arduino\libraries\SparkFun_Simultaneous_RFID_Tag_Reader_Library\src\SparkFun_UHF_RFID_Reader.cpp: In member function 'void RFID::printMessageArray()':
c:\Users\<username>\Documents\Arduino\libraries\SparkFun_Simultaneous_RFID_Tag_Reader_Library\src\SparkFun_UHF_RFID_Reader.cpp:891:20: warning: comparison is always false due to limited range of data type [-Wtype-limits]
     if (amtToPrint > MAX_MSG_SIZE)
c:\Users\<username>\Documents\Arduino\libraries\SparkFun_Simultaneous_RFID_Tag_Reader_Library\src\SparkFun_UHF_RFID_Reader.cpp: In member function 'uint8_t RFID::parseResponse()':
c:\Users\<username>\Documents\Arduino\libraries\SparkFun_Simultaneous_RFID_Tag_Reader_Library\src\SparkFun_UHF_RFID_Reader.cpp:754:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
Compiling core...
```

---

I'll pass this issue over to the engineering team to test the [EspSoftwareSerial library](https://github.com/plerup/espsoftwareserial), add it as a dependecy for Espressif boards, and fix the return mismatch.

**Notes:**

- Currently, the ESP32 Arduino core doesn't support software serial because a hardware UART can be attached to pretty much any pin. That pretty much breaks most of the examples, when the library is used with an ESP32 board.
- Additionally, the [EspSoftwareSerial library](https://github.com/plerup/espsoftwareserial) recently added to the ESP8266 Arduino core
  - Both could be addressed by using the [EspSoftwareSerial library](https://github.com/plerup/espsoftwareserial). Would need to doublecheck for compiler issues and test the functionality
- There is a return mismatch, it looks like hardware/software libraries of the Arduino cores no longer return anything when they are initialized.
- If fixed, the `README` and hookup guide would need to be updated.
  - Thing Plus vs R3 pin layout
  - Caution note for connecting the shield to 3.3V microcontrollers
    - Logic level adapter
    - Powering the module (3.7-5.5V)
  - Add EspSoftwareSerial library dependency

closed 2023-03-14 by sfe-SparkFro